sysprog21/lkmpg #4: Design new book covers

jserv, 2021-07-23:
It has been 16 years since last book cover for "[The Linux Kernel Module Programming Guide](https://tldp.org/LDP/lkmpg/2.6/lkmpg.pdf)". We would change the cover sooner or later.

Candidate cover:
[image]

bitter-bits, 2021-07-24:
This one is a similiar design, with unified font type and new house:
[image]

bitter-bits, 2021-07-24:
This one is a similiar design, with unified font type and new house:
[image]

[Full size](https://raw.githubusercontent.com/bitter-bits/linux2021-quiz/main/2021-07-24_150546.png)

jserv, 2022-01-09:
Thank @bitter-bits and @jkrvivian for contributing! The cover page is landed.

closed 2022-01-09 by jserv