sysprog21/lkmpg #5: Introduce User Mode Linux?

ekangmonyet, 2021-07-23:
Is it beneficial to introduce UML as an alternative to running linux VM under linux?
**PROS**
- Easy access to hostfs
- Lightweight
- Debuggable from host (attach to gdb)

**CONS**
- Little documentation
- I don't know enough of it ='(

ekangmonyet, 2021-07-23:
**Demonstration:**
Inserting hello world module on 5.12.19-usermode, however it doesn't show up in lsmod and can't be removed for some yet unknown reasons.
[image]

jserv, 2021-07-23:
> **Demonstration:**
> > ./linux init=/bin/bash rootfstype=hostfs
> 
> Inserting hello world module on 5.12.19-usermode, however it doesn't show up in lsmod and can't be removed for some yet unknown reasons.

Did you follow the build instructions? https://hackmd.io/@sysprog/user-mode-linux-env
You have to specify `ARCH=um` at compile time.

In addition, don't put the pictures which consist of command line messages only. Instead, always use GitHub-flavored Markdown syntax for code.

ekangmonyet, 2021-07-23:
I didn't know that exist!  Thanks to that I now have a better understanding on the whole UML thing.
I was using my host root as the rootfs and thus /proc/ refers to my host /proc/ which it should not.
Then I created an [alpine rootfs](https://alpinelinux.org/downloads/), mounted proc on /proc, everything seems working fine:
```
/hello-1 # insmod hello-1.ko
HELLO WORLD 2.
/hello-1 # lsmod
Module                  Size  Used by    Tainted: G
hello_1                  872  0
/hello-1 # rmmod hello-1.ko
GOODBYTE WORLD 1.
/hello-1 #
```
Should we try to add a section about setting up UML?  I can start writing about that and see how things go.

jserv, 2021-07-24:
> Should we try to add a section about setting up UML? I can start writing about that and see how things go.

Instead of UML, I would like to use QEMU for device driver development for the sake of peripheral support. UML is perfect for people who are hacking the internals of Linux kernel, but it is quite limited for device. There is single CPU supported in UML as well.

ekangmonyet, 2021-07-24:
Ok, that's true.  I've gotten QEMU to work too.  Let's close this first then.

closed 2021-07-24 by ekangmonyet