sysprog21/lkmpg #25: Use GitHub Actions for site building and format conversion.

Hsins, 2021-07-31:
There are a bunch of [static site generator](https://jamstack.org/generators/) that creates HTML pages from templates or components and the given content source. Besides, we can also automate, customize, and execute workflows right in the repository with [GitHub Actions](https://github.com/features/actions).

The [OI Wiki Project](https://github.com/OI-wiki/OI-wiki) is an example:

1. The contents are written in Markdown.
2. Then the `build` and `build-pdf` workflows are triggered whenever there is any push or pull request event.
    - `build` workflow runs [`mkdocs`](https://www.mkdocs.org/) for creating HTML pages from `*.md` files.
    - `build-pdf` workflow coverted `*.md` files to `*.tex` files and compiled them to `output.pdf`.

Besides, the site [《大家來學 LaTeX》](https://textw.github.io/LaTeX123/) is another site created with [Hugo](https://gohugo.io/) and use [utteranc](https://utteranc.es/) as the comment system. It would be a possible solution to use [pandoc](https://pandoc.org/demos.html) for coverting `*.tex` to `*.md`.

jserv, 2021-07-31:
make4ht is used for HTML generation (#20), and I would like to LaTeX script at present.

Hsins, 2021-07-31:
Sure that we can keep the source files in `*.tex` format but find out another solution for HTML/EPUB generating. It's not really convenient to [use `TeX4ht` with custom HTML template](https://tex.stackexchange.com/questions/561450/how-to-use-my-own-html-template-with-make4ht) but it's enough at present.

fennecJ, 2021-07-31:
I think build a markdown file then convert it into html sounds promising.
However `pandoc` doesn't support some extensions used in `lkmpg.tex` and caused several code blocks to be omitted.

Hsins, 2021-07-31:
More packages make more possible dependency issue but we don't have bunch of packages. 

I've check that currently the format conversion of code block from `tex` to `md` would be like something below:

<table>
<tr>
  <td><b>LaTeX</b></td>
  <td><b>Markdown</b></td>
</tr>

<tr>
  <td>

```plain
\subsection{Kernel module package}
\label{sec:packages}

Linux distributions provide the commands 
\emph{modprobe}, \emph{insmod} and \emph{depmod} 
within a package.

On Ubuntu/Debian:
\begin{codebash}
sudo apt-get install build-essential kmod
\end{codebash}
```

  </td>
  <td>

```plain
## Kernel module package {#sec:packages}

Linux distributions provide the commands 
*modprobe*, *insmod* and *depmod* 
within a package.

On Ubuntu/Debian:

::: codebash
sudo apt-get install build-essential kmod
:::
```
  </td>
</tr>
</table>

It seems that we can do something tricky: `TeX -> Markdown -> HTML`. Just replace `":::"` with `"```"` and `"codebash"` (or something else programming languages) with `"bash"` before converting to `HTML` from `Markdown`.

Hsins, 2021-07-31:
Also, I found that we can use `listings` package instead of `minted` in LaTeX and `pandoc` handle the `lstlisting` environment well.

[image]

Hsins, 2021-07-31:
After running with a custom [pandoc filter](https://github.com/jgm/pandocfilters) for `minted` package. Now it can handle the code block environment with `minted` package well:

[image]

It may take me some time for creating the pandoc filter to deal with the `\newminted[codebash]` environment that @jserv use.

fennecJ, 2021-08-01:
Consider using `\begin{minted}[linenos]{c}` instead of `\begin{minted}{c}` so the line number can be shown.

Hsins, 2021-08-04:
There is a POC (Proof of Concept) site building with `Pandoc + Hugo (theme: book)` could be check at [here](https://textw.github.io/lkmpg/docs/lkmpg/). It seems that the lua-filters `minted` given by Pandoc offically could not deal with the `[linenos]` options well. And I need to modified for custom usage. 

Now, waiting for #26 resolved so that I could split one page into several chapters.

jserv, 2026-04-24:
> There is a POC (Proof of Concept) site building with `Pandoc + Hugo (theme: book)` could be check at [here](https://textw.github.io/lkmpg/docs/lkmpg/). It seems that:
[...]
> Now, waiting for [#26](https://github.com/sysprog21/lkmpg/issues/26) resolved so that I could split one page into several chapters.

@Hsins , can you help review recent LKMPG?