sysprog21/lkmpg #26: Modular LaTeX Documents.

Hsins, 2021-07-31:
As the work grows, the `*.tex` file can become unwieldy and confusing, especially we're writing a full-length book. It's the good practice to split the file into several files.

LaTeX makes this very easy with the commands `\input{filename}` and `\include{filename}`. A template for the modular latex document can be set up as follows structure:

```
.
├── back
│   ├── appendix-*.tex
│   ├── references.bib
│   └── ...
├── contents
│   ├── chapter-*.tex
│   └── ...
├── figures
│   ├── chapter-*
│   │   ├── *.[png/jpg/pdf]
│   │   └── ...
│   └── ...
├── main.tex
├── README.md
├── [style.tex]
├── [template.cls]
├── ...
├── LICENSE
├── Makefile
└── README.md
```

## References

- [Modular LaTeX Documents | Olivier Pieters](https://olivierpieters.be/blog/2015/12/04/modular-latex-documents)
- [Modular Latex | Eitan Lees](https://eitanlees.github.io/latex/2019/05/17/modular-latex.html)
- [NTU-Thesis-LaTeX-Template | GitHub](https://github.com/Hsins/NTU-Thesis-LaTeX-Template)

Hsins, 2021-08-01:
I've split the files following the structure above and connect it with Overleaf [here](https://www.overleaf.com/read/ckxnwfcprbwk). If @jserv has no problem with it then I'll then commit and make a pull request for this.

Besides, why do you use the documentclass `book` but use `\section{}` insteaf of `\chapter{}`? Is there any concern?

jserv, 2021-08-01:
> Besides, why do you use the documentclass `book` but use `\section{}` insteaf of `\chapter{}`? Is there any concern?

The TeX script was converted from [Org mode for Emacs](https://orgmode.org/), and you can figure out some kludge.

jserv, 2021-08-01:
Due to historical reasons, LKMPG was poor at its documentation organization. That is, the sections/chapters were tattered from the perspective of a complete programming guide. Before the modularization, it would be better if we can revise the existing materials and amalgamate parts of the sections. Recently, I have been preparing.

jserv, 2025-10-31:
> you started this over four years ago and blocks #59. Is this work still active?

Unfortunately, it appears to be stuck. If you are interested in taking on this task, please propose a reorganization plan for each chapter.

closed 2026-04-23 by jserv