sysprog21/lkmpg #27: HTML publishing using Github Actions and Docker

ekangmonyet, 2021-08-01:
This is to centralize efforts of developing Github Actions and **_if needed_** a customized docker container for the purpose of building the HTMLs.

ekangmonyet, 2021-08-01:
I think there's a cyclic dependency between this and #20 , my opinion is to merge #20 first, start this and improve HTML building at the same time.

Hsins, 2021-08-01:
Just focus on "building docker image for consistent environment" and then the issue could be the first task to be done. The title of this issue are almost the same with #25.

ekangmonyet, 2021-08-01:
I will focus on producing the PDF file first instead, since it is pretty stable and no other task dependencies. It will serve as a foundation to build other assets too.

fennecJ, 2021-08-01:
For html building, the following script worked for me:
``` yml
name: Generate HTML and deploy to GitHub Page.

on:
  push:
    branches:
      - master

jobs:
  build-deploy:
    runs-on: ubuntu-latest
    steps:
      - name: checkout branch
        uses: actions/checkout@v2

      - name: install dependencies and generating files
        uses: xu-cheng/texlive-action/full@v1
        with:
          run: |
            apk add make
            make html

      - name: deploy site
        uses: peaceiris/actions-gh-pages@v3
        with:
          github_token: ${{ secrets.GITHUB_TOKEN }}
          publish_dir: ./html                                          # --> should be the same as your output folder of make4ht
          publish_branch: gh-pages
```
And you can checkout the output in [here](https://fennecj.github.io/lkmpg/)
Looks like the page rendered well.

ekangmonyet, 2021-08-02:
#32 is working on it.

Hsins, 2021-08-02:
@fennecJ @ekangmonyet 

Currently both workflow [texlive-action](https://github.com/xu-cheng/texlive-action) and container [texlive](https://hub.docker.com/r/texlive/texlive) are build for general use, and they're not actually an "reproducible TeX build environment" for our project.

I've build [this docker image](https://github.com/TeXtw/docker-lkmpg) supports full TeXLive with Gnuplot and Pandoc. If you have installed docker on your local machine, try with following steps to compile with it:

```bash
# clone the source code
$ git clone https://github.com/sysprog21/lkmpg && cd lkmpg

# pull docker image and run it as container
$ docker pull twtug/lkmpg
$ docker run --rm -it -v $(pwd):/workdir twtug/lkmpg

# run commands to compile documents
$ pdflatex -shell-escap lkmpg.tex
$ make4ht --shell-escape --utf8 --format html5 --config html.cfg --output-dir html lkmpg.tex
```

Hsins, 2021-08-02:
Hi @jserv, could you check this [Dockerfile](https://github.com/TeXtw/docker-lkmpg/blob/main/Dockerfile) and tell me any packages should be add or remove? Moreover, do you prefer any other docker registries (Docker Hub/GitHub Packages...etc) for storing the docker image?

jserv, 2021-08-02:
>  could you check this [Dockerfile](https://github.com/TeXtw/docker-lkmpg/blob/main/Dockerfile) and tell me any packages should be add or remove? Moreover, do you prefer any other docker registries (Docker Hub/GitHub Packages...etc) for storing the docker image?

I am no expert in Docker. It seems to be comprehensive to me.

Hsins, 2021-08-02:
> I am no expert in Docker. It seems to be comprehensive to me.

Just consider that docker is something like virtual machine but much lighter. Dockers based on container technology and the containers are just user space of the operating system.

Let me try to make a long story short: 

- The **Docker Container** is a runnable instance of a **Docker Image**.
- The **Dockerfile** is a text-based script of instructions used to create a **Docker Image**.
- The **Docker Registry** is somewhere to storing **Docker Images**.
- Moreover, tasks in GitHub Actions workflows could be executed on given Docker Image.

Actually, you can just tell me the packages and software to be installed for building this project.

ekangmonyet, 2021-08-02:
> Actually, you can just tell me the packages and software to be installed for building this project.

I think it's working quite fine for now we can leave it as is, until we encounter a solid problem.  What do you think?

Hsins, 2021-08-04:
> I think it's working quite fine for now, we can leave it as is until we encounter a solid problem. What do you think?

Sure!

closed 2021-08-04 by ekangmonyet