sysprog21/lkmpg #35: Device-tree support

cruise-nclark, 2021-08-02:
Would it be possible to get a section that talks about device-trees and how they interact with kernel modules? That would be a great addition for the folks that work on embedded systems.

jserv, 2021-09-08:
@johnnylord, can you attempt to contribute preliminary descriptions along with sample code for device tree?
* https://github.com/johnnylord/drivers/tree/main/06-devicemodel
* https://github.com/johnnylord/drivers/tree/main/07-platform

johnnylord, 2021-09-09:
I will try to give the device tree example with raspberry pi (the open source one). As device tree is proposed to decouple the hardware description from the kernel source tree, some companies maintain their own device tree files for different platforms and SoCs product internally, and their device tree files are not public available.

jserv, 2025-04-12:
> I will try to give the device tree example with raspberry pi (the open source one). As device tree is proposed to decouple the hardware description from the kernel source tree, some companies [...]

@johnnylord, feel free to submit pull requests here.

commit f67fb830 (2025-08-06): Add Device Tree section

This commit adds 'devicetree.c' and 'dt-overlay.dts', demonstrating the
platform driver and DT interaction. Also, it adds comprehensive Device
Tree section to documentation covering:
* Introduction to Device Tree concepts
* Integration with kernel modules
* Property reading examples
* Testing methods and common functions

Close #35
commit 06cd0f4c (2025-08-08): Add Device Tree section

This commit adds 'devicetree.c' and 'dt-overlay.dts', demonstrating the
platform driver and DT interaction. Also, it adds comprehensive Device
Tree section to documentation covering:
* Introduction to Device Tree concepts
* Integration with kernel modules
* Property reading examples
* Testing methods and common functions

Close #35