sysprog21/lkmpg #36: Text does not scale on firefox

nowakf, 2021-08-02:
Book text doesn't scale in Firefox 90.0.2, also chromium. It makes it pretty hard to read if you have bad eyesight.

ekangmonyet, 2021-08-03:
I would advocate for [Firefox Reader View](https://support.mozilla.org/en-US/kb/firefox-reader-view-clutter-free-web-pages).  If the reflow is accurate and stable, it can also be released as epub format.

fennecJ, 2021-08-04:
I think the reason why text can't be scale is because in `html.cfg`, line 13,
```css
font-size: 1vw;
```
When an element using `vw` as size unit, its size will be depended on the
width of the browser window size and won't be effected by scaling ratio
of browser. I tried to use `em` or `px` instead, however when scaling the
text, the code blocks' layout will be malformed and I currently can't find
a way to fix that.

jserv, 2021-08-05:
Close via commit a411938

closed 2021-08-05 by jserv