sysprog21/lkmpg #40: Update Document References

dsrm23, 2021-08-04:
While looking at the examples and attempting to lookup the reference docs that are mentioned, I have been finding it difficult to track down these docs. A particular example is "linux/drivers/char/Makefile" referenced in section 0.4.2. I am unable to find any reference to this location. Is this an anachronism?

In general the reference style is provides a relative link, but does not give a point of reference from the filesystem. This could easily be fixed via actual HTML links or by including instructions to also install the relevant kernel doc packages in the setup guide early in the doc along with a sentence providing a point of reference. 

Thanks for the resource.

Hsins, 2021-08-05:
Currently, the HTML file is converted from `*.tex` source via [make4ht](https://github.com/michal-h21/make4ht) so it's not a good practice to modified HTML file directly.

This should be done with following replacement when writing `.*tex` source file instead:

```tex
- Now have a look at \verb|linux/drivers/char/Makefile| for a real world example.
+ Now have a look at \href{https://github.com/torvalds/linux/blob/master/drivers/char/Makefile}{\verb|linux/drivers/char/Makefile|} for a real world example.
```

jserv, 2021-08-09:
The latest git commit introduces new command `\src` for preliminary hyperlink generation which points to Linux kernel source code. At present, stable/linux.git was referred: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git

jserv, 2021-08-09:
Close via commit 2d368210cef20c620007839497285bc50d0ceb76

closed 2021-08-09 by jserv