sysprog21/lkmpg #59: A better online version is published

holmescn, 2021-08-14:
Nice work for all the writers. Since I am going to write some Linux Kernel Module, I found this new book that explain the new infrastructure in modern Linux Kernel. But I found that the online version feels not good enough. So I spend a week to convert the latex content into a simple yet I think it is better mark language [https://asciidoc.org](AsciiDoc). The project is based on [https://antora.org](Antora) which is a new writing platform similar to Sphinx. [https://matthewsetter.com/why-antora-is-the-leading-technical-writing-platform/](Here) is an article that explains why I chosen it.

* [http://www.holmesconan.me/lkmpg/](Github Pages)
* [https://github.com/holmescn/lkmpg](Github Repo)

The content is separated into multiple files which I think is easier to maintain. I hope the latex could also be separated into chapters if you like. But there is a problem that the Github Pages published under my personal domain, I don't known if you like to merge it, or create an organization that place this version under it.
Another problem is that current version of Antora could not generate PDF files, so the latex version could not be replaced. But the underhood AsciiDoc is targeted DocBook whichi could be converted to PDF. So I think this could be resolved later.

jserv, 2021-08-14:
Thank @holmescn for working on Antor based reformatting. I would like to consult @fennecJ and @Hsins .

Hsins, 2021-08-14:
Hi @holmescn, I remember [@jserv mentioning that he would like to stick to the LaTeX scripts at present](https://github.com/sysprog21/lkmpg/issues/25#issuecomment-890374742). Based on that principle, **the LaTeX scripts should not be replaced** even though the PDF files could be generated with Antora, but it would be nice to generate different PDF files based on both and then bundle them in release for choosing.

That is to say, all the other formats (HTML, epub, mobi, and so on) should be converted from LaTeX files automatically as possible as it could and we should only edit the LaTeX files for unifying our work. Could you offer the **process** for converting LaTeX to AsciiDoc format? I think a possible workflow would be like the diagram shown below:

<p align="center">
  [image]
</p>

<p align="center">
  (Note that the LaTeX files should be the source files.)
</p>

There are three (or more with other tools) approach to handle that. I believe that both AsciiDoc and Markdown are brilliant markup document format for us because we won't have too much modern features to make this book great. Before we decide which workflow to be used, the issue #26 should be resolve first.

> As a reminder, some of custom macros like [this](https://github.com/sysprog21/lkmpg/blob/2d368210cef20c620007839497285bc50d0ceb76/lib/kernelsrc.tex) should be prevent to simplify those workflows. The more custom macros we use, the less scalability we have. Only the `make4ht` handle those macros well because it totally based on LaTeX engine.

fennecJ, 2021-08-15:
I agree with @Hsins 
We should only modify latex file and leave other files to be generated automatically.

holmescn, 2021-08-16:
@Hsins sorry for that in current stage, the AsciiDoc is not fully automatically generated. I just written a simple Python script to split the `lkmpg.tex` into several tex files and then convert it manually. There are some problem with pandoc since the `\sample` macro is not convertable. And there is also a diagram that I didn't converted. I will do my best to write a convert script, but I have no idea about the deadline. I thought this issue could be closed until I finished the script.

Hsins, 2021-08-16:
@holmescn Almost the custom macros like `\sample` and `\src` are not convertable with pandoc. Just take a break and wait for the chapter splitting issue #26 to be done.