sysprog21/lkmpg #61: MODULE_SUPPORTED_DEVICE removed

nlantau, 2021-08-16:
In section [0.4.4 Licensing and Module Documentation](https://sysprog21.github.io/lkmpg/#licensing-and-module-documentation) the code of **hello-4.c** uses 
```c
MODULE_LICENSE("GPL"); 
MODULE_AUTHOR("LKMPG"); 
MODULE_DESCRIPTION("A sample driver"); 
MODULE_SUPPORTED_DEVICE("testdevice"); 
```
where **MODULE_SUPPORTED_DEVICE** is remove from upstream since [March 17, 2021](https://github.com/torvalds/linux/commit/6417f03132a6952cd17ddd8eaddbac92b61b17e0#diff-1a8b5118e6eaf5be44fdf51f9bceaf816aec6d96e4a048b8455bcfbca6a680f2).

jserv, 2021-08-16:
Thank @nlantau for pointing this out!

closed 2021-08-16 by jserv

nlantau, 2021-08-16:
> Thank @nlantau for pointing this out!

No worries! Thanks for a great guide!

commit b497b6a3 (2021-08-16): Remove never implemented MODULE_SUPPORTED_DEVICE

MODULE_SUPPORTED_DEVICE is remove from upstream since March 17, 2021.
See linux.git commit 6417f03

Reported by Niklas Lantau <niklaslantau@gmail.com>
Close #61