sysprog21/lkmpg #64: Mention PDF download information in generated HTML

jserv, 2021-08-16:
At present, HTML is rendered via `make4ht`. However, it lacks the information about PDF downloading.

The expected output:
1. Change the HTML template and/or setting to add the PDF download hyperlink.
2. Mention PDF download only in generated HTML. The body of PDF document should not contain the download link though.

fennecJ, 2021-08-17:
I come up with 2 solutions for this issue
* 1 We can add
  ```
  \begin{center}
  \href{https://github.com/sysprog21/lkmpg/releases/download/latest/lkmpg.pdf}{Download this page in pdf}
  \end{center}
  ```
  At the end of `html.cfg`, and the information about PDF downloading can only be at the top of the page:
  [image]

* 2 Using \ifdefined\HCode
  By this way we can choose where to place the information about PDF downloading.
  [image]

jserv, 2021-08-17:
How about following [Yocto Project Overview and Concepts Manual](http://docs.yoctoproject.org/overview-manual/)? We can put the hyperlinks to PDF download and GitHub  in the upper right corner of the HTML page.

fennecJ, 2021-08-17:
Add following code
```diff
--- a/html.cfg
+++ b/html.cfg
@@ -45,5 +45,15 @@
     user-select: none;
 }}
 
+\Css{.flushright:first-child {
+    position:absolute;
+    top:0;
+    right:0;
+}}
+
 \begin{document}
 \EndPreamble
+
+\begin{flushright}
+\href{https://github.com/sysprog21/lkmpg/releases/download/latest/lkmpg.pdf}{Download this page in pdf}
+\end{flushright}
```
into `html.cfg` would be looked like:
[image]

You can check it in [https://fennecj.github.io/lkmpg/](https://fennecj.github.io/lkmpg/)

jserv, 2021-08-17:
Change from "Download this page in pdf" to "Download PDF document"

jserv, 2021-08-17:
> You can check it in https://fennecj.github.io/lkmpg/

Can you reserve some spaces at the end of the line "Download PDF document"?

fennecJ, 2021-08-17:
Did you mean something looks like this?
[image]
You can modify the space between word and right side through `right` value, and modifiy the space between word and top side through `top` value.
```diff
+\Css{.flushright:first-child {
+    position:absolute;
+    top: 10px;
+    right: 50px;
+}}
+
```

jserv, 2021-08-17:
> You can modify the space between word and right side through `right` value, and modifiy the space between word and top side through `top` value.

It looks great to me. Please send the corresponding pull request.

commit fe8dd00e (2021-08-17): Add PDF download link in html only

Close #64
closed 2021-08-17 by jserv