sysprog21/lkmpg #68: Update Acknowledgements

jserv, 2021-08-17:
The current "Acknowledgements" section is not up to date. We can generate the new list of contributions by looking up git log. The command `git shortlog examples lkmpg.tex` works out of the box.

However, we might need extra table to convert GitHub account into the names in Pinyin.

jserv, 2021-09-06:
A good example is [The Not So Short Introduction to LATEX 2ε](https://tobi.oetiker.ch/lshort/lshort.pdf)
source: https://github.com/oetiker/lshort/blob/master/book/src/contrib.tex

jserv, 2021-09-25:
Since we have received the contributions from individuals, it is time to update acknowledgement list. Preliminary steps:
1. Create a new TeX document, `contrib.tex`, which preserves the names and email address in alphabetical order.
2. Include `contrib.tex` in section "Acknowledgements"
3. Ensure each contributor is listed within compact layout.

fennecJ, 2021-09-26:
I manage to write a [script](https://github.com/fennecJ/lkmpg/blob/master/.ci/gen-contributor-list.sh) to generate the contributors' list and parse the output into [contrib.tex](https://github.com/fennecJ/lkmpg/blob/master/lib/contrib.tex)
However, some contributors are duplicated due to different email addresses they use:
[image]

This can be solved via configuring `.mailmap` manually, but I don't know which email is their main email.

Currently the output can be checked in here:
https://fennecj.github.io/lkmpg/#acknowledgements

linD026, 2021-09-26:
We can open the new issue or use the current one to tell the contributor to give us their main email.
By the way, my main email is `shiyn.lin@gmail.com`. And the `linzhien` is also my sub-account.
Thanks.

jserv, 2021-09-26:
> I manage to write a [script](https://github.com/fennecJ/lkmpg/blob/master/.ci/gen-contributor-list.sh) to generate the contributors' list and parse the output into [...]

You can prepare the script(s) in directory `scripts/`, which is intended for maintenance purpose. The format of contributor list can be as following:
# Name (to be listed in LKMPG) # Hanyu Pinyin # email-1 # email-2 # remark

Take me for example,
```
Jim Huang, Ching-Chun Huang, jserv.tw@gmail.com, jserv@ccns.ncku.edu.tw, jserv@biilabs.io
```

Then, we can make use of `grep` to see if the contributor was already listed or not.

jserv, 2021-09-26:
> You can prepare the script(s) in directory `scripts/`, which is intended for maintenance purpose. The format of contributor list can be as following:
>
> ```
> Name (to be listed in LKMPG), Hanyu Pinyin, [...]

jserv, 2021-09-26:
> This can be solved via configuring .mailmap manually, but I don't know which email is their main email.

Only one name would appear in the Acknowledgements  section of LKMPG. No email address is shown except me (as the maintainer).

fennecJ, 2021-09-27:
> Only the preferable name would appear in the Acknowledgements section of LKMPG. No email address is shown except me (as the maintainer).

I know, the reason why I wanted to know the main mail of contributors is that I want to maintain the list as the example source 
code you mentioned before (https://github.com/oetiker/lshort/blob/master/book/src/contrib.tex).
Which is in the format of
Name,       % Email
Where the % causes latex ignore the info after it.

I can also make scripts to generate the list as you said, that is
```
Name (to be listed in LKMPG)[, Hanyu Pinyin], email-1[, email-2][, email 3][, remark (usually GitHub account)]
```
Which puts different emails in the same line if they belong to the same name.
Then use another script to parse the list and generate the `contrib.tex`.
But I wonder how will we deal with the situation that two different contributors use the same name in commit logs.
If that thing happened, will we put their emails in the same line or another new line?

jserv, 2021-09-27:
> But I wonder how will we deal with the situation that two different contributors use the same name in commit logs.
> If that thing happened, will we put their emails in the same line or another new line?

It implies that we have much stronger community at that time. We can then manipulate manually.

commit af09a8a1 (2021-11-08): Add script to gen contributor list and contrib.tex

This script do mainly 2 things:
1.
Generate file `Contributors` with git log in following format:
Contributor's name,<1-st e-mail>[,<2nd e-mail>][,<3rd e-mail>]...
The 2-nd email and so on are based on file `.mailmap` in the root
directory of the repo.
Note that it will also append contributors in File `Include`; and
will NOT append contributors in File `Exclude`.
If there are new contributors, script will sort `Contributor` after
append new contributors into each file; Otherwise it do nothing.

2.
Generate contrib.tex based on `Contributors` [... cut]
closed 2021-11-08 by jserv