sysprog21/lkmpg #72: Set up CI infrastructure to automate Linux kernel module validation

jserv, 2021-08-23:
At present, CI only checks the coding style. It is far from complete validation. We shall set up the CI infrastructure which is able to perform:
1. Build all example Linux modules within Ubuntu Linux 20.04 (kernel 5.4+);
2. Attempt to do "insmod" and "rmmod";
3. (Optionally) check the kernel messages and validate;

See https://github.com/kernelci/kernelci-core as an example.

fennecJ, 2021-08-25:
I wrote a simple shell script trying "insmod" and "rmmod" all module on my machine and meet following errors:
`chardev2.ko`:
At first time insmod and rmmod didn't throw any errors, however when using command `dmesg`, it shows:
```bash
[   78.479322] chardev2: loading out-of-tree module taints kernel.
[   78.479463] chardev2: module verification failed: signature and/or required key missing - tainting kernel
[   78.480170] Device created on /dev/char_dev
```
And when I tried to insmod the module again, terminal shows:
```bash
insmod: ERROR: could not insert module chardev2.ko: Device or resource busy
```
And the output of `dmesg` was:
```bash
[  157.168167] Sorry, registering the character device  failed with -16
```
---
`bottomhalf.ko`:
`intrpt.ko`:
When insmod `bottomhalf.ko` or `intrpt.ko`, the error was something like:
```bash
insmod: ERROR: could not insert module xxx.ko: Unknown error 517
```
And the output of `dmesg` was:
```bash
Unable to request GPIOs for LEDs: -517
```
I think my board's GPIO number may be different from sample code, which cause this error.
But I currently don't know how to check GPIO number of my board.

---
`syscall.ko`
When insmod `syscall.ko` first time, the terminal simply shows
```bash
killed
```
And the output of `dmesg` was:
```bash
[  384.864739] BUG: unable to handle page fault for address: ffff8aad47458000
[  384.864755] #PF: supervisor read access in kernel mode
[  384.864757] #PF: error_code(0x0000) - not-present page
[  384.864758] PGD 1aa01067 P4D 1aa01067 PUD 4907ff067 PMD 4907fe067 PTE 0
[  384.864761] Oops: 0000 [#1] SMP NOPTI
[  384.864763] CPU: 3 PID: 3017 Comm: insmod Tainted: G           OE     5.11.0-27-generic #29~20.04.1-Ubuntu
[  384.864765] Hardware name: CJSCOPE INC. Z Series/Z Series, BIOS 1.07.07TCOP1 05/09/2019
[  384.864766] RIP: 0010:syscall_start+0x15/0x1000 [syscall]
[  384.864771] Code: Unable to access opcode bytes at RIP 0xffffffffc056bfeb.
[  384.864772] RSP: 0018:ffff9ae78084bc80 EFLAGS: 00010293
[  384.864774] RAX: ffff8aad47457fe8 RBX: 0000000000000000 RCX: 0000000000000000
[  384.864775] RDX: ffff8aad47457fe8 RSI: ffffffff9b75cea8 RDI: ffffffffc056c000
[  384.864776] RBP: ffff9ae78084bce8 R08: 0000000000000010 R09: ffff8aae578d7370
[  384.864777] R10: ffff8ab1907d4b80 R11: 0000000000000000 R12: ffffffffc056c000
[  384.864778] R13: ffff8aae578d7370 R14: 0000000000000000 R15: ffff9ae78084be70
[  384.864779] FS:  00007f60559ef540(0000) GS:ffff8ab1806c0000(0000) knlGS:0000000000000000
[  384.864781] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  384.864782] CR2: ffffffffc056bfeb CR3: 0000000109480002 CR4: 00000000003706e0
[  384.864783] Call Trace:
[  384.864785]  ? do_one_initcall+0x48/0x1d0
[  384.864789]  ? _cond_resched+0x19/0x30
[  384.864792]  ? kmem_cache_alloc_trace+0x380/0x430
[  384.864794]  ? do_init_module+0x28/0x250
[  384.864797]  do_init_module+0x62/0x250
[  384.864799]  load_module+0x11aa/0x1370
[  384.864802]  ? security_kernel_post_read_file+0x5c/0x70
[  384.864804]  ? security_kernel_post_read_file+0x5c/0x70
[  384.864807]  __do_sys_finit_module+0xc2/0x120
[  384.864809]  ? __do_sys_finit_module+0xc2/0x120
[  384.864811]  __x64_sys_finit_module+0x1a/0x20
[  384.864813]  do_syscall_64+0x38/0x90
[  384.864816]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[... 14 lines cut ...]
[  385.030385] RSP: 0018:ffff9ae78084bc80 EFLAGS: 00010293
[  385.030387] RAX: ffff8aad47457fe8 RBX: 0000000000000000 RCX: 0000000000000000
[  385.030388] RDX: ffff8aad47457fe8 RSI: ffffffff9b75cea8 RDI: ffffffffc056c000
[  385.030389] RBP: ffff9ae78084bce8 R08: 0000000000000010 R09: ffff8aae578d7370
[  385.030390] R10: ffff8ab1907d4b80 R11: 0000000000000000 R12: ffffffffc056c000
[  385.030391] R13: ffff8aae578d7370 R14: 0000000000000000 R15: ffff9ae78084be70
[  385.030392] FS:  00007f60559ef540(0000) GS:ffff8ab1806c0000(0000) knlGS:0000000000000000
[  385.030394] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  385.030395] CR2: ffffffffc056bfeb CR3: 0000000109480002 CR4: 00000000003706e0
```
And when trying to insmod `syscall.ko` again, terminal just freeze without any output.

---
`start.ko`:
insmod seems worked well.
And no error message found in `dmesg`:
```bash
Hello, world - this is the kernel speaking
```
However when trying to rmmod, terminal showed:
```bash
rmmod: ERROR: ../libkmod/libkmod-module.c:799 kmod_module_remove_module() could not remove 'start': Device or resource busy
rmmod: ERROR: could not remove module start: Device or resource busy
```
Perhaps we should fix these errors before we set up CI flows?

jserv, 2021-08-25:
Both `intrpt` and `bottomhalf` were dedicated to Raspberry Pi. It is expectedly not working on x86-based VM instance.

jserv, 2021-08-25:
@linD026, can you help resolve the LKM problems?

linD026, 2021-08-25:
In Multipass(5.4.0-80-generic), I also got the error messages:
```shell
[   65.668320] chardev2: loading out-of-tree module taints kernel.
[   65.668338] chardev2: module verification failed: signature and/or required key missing - tainting kernel
[   65.670278] Device created on /dev/char_dev
```

Checking out this [guide](https://www.kernel.org/doc/html/latest/admin-guide/tainted-kernels.html) found out:

```shell
ubuntu@ubuntu:~/lkmpg/examples$ sudo insmod chardev2.ko
ubuntu@ubuntu:~/lkmpg/examples$  cat /proc/sys/kernel/tainted
12288
ubuntu@ubuntu:~/lkmpg/examples$  for i in $(seq 18); do echo $(($i-1)) $(($(cat /proc/sys/kernel/tainted)>>($i-1)&1));done
0 0
1 0
2 0
3 0
4 0
5 0
6 0
7 0
8 0
9 0
10 0
11 0
12 1
13 1
14 0
15 0
16 0
17 0
```

> 12 O if an externally-built (“out-of-tree”) module has been loaded.
> 13 E if an unsigned module has been loaded in a kernel supporting module signature.

I still looking at why this happened.

linD026, 2021-08-31:
The problems of `chardev2.ko` and `syscall.ko` have been fixed.
See the commit [06b7594](https://github.com/sysprog21/lkmpg/commit/06b75942ccd03d61e150dfc9e57dc12757a5b777) and [cccc98a](https://github.com/sysprog21/lkmpg/commit/cccc98ab2c914cc922d34266b88ff73231e06607).

commit 3fcbb8e1 (2021-09-08): Improve CI pipeline to automate module testing (#95)

The script executes insmod/rmmod twice with available kernel modules.

Close #72
closed 2021-09-08 by jserv