sysprog21/lkmpg #87: Follow Linux kernel coding style

jserv, 2021-09-01:
We should follow [Linux kernel coding style](https://www.kernel.org/doc/html/latest/process/coding-style.html) as possible. The major difference in LKMPG is to use 4 spaces for indention rather than tabs. This is because we would stick to compact layout for source listing. However, others rules should be enforced. e.g., CAPITALIZED macro names are appreciated but macros resembling functions may be named in lower case. There are some example files that violates the rule, such as `examples/chardev2.c`. We should replace the use of [CamelCase](https://en.wikipedia.org/wiki/Camel_case) with plain lowercase variables.

linD026, 2021-09-02:
I found the [clang-format](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/.clang-format?h=v5.14) file in Linux.
The explaination is in [/Documentation/process/clang-format.rst](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/process/clang-format.rst?h=v5.14).
We can use it, so I adjust a little bit:
```
Language: Cpp

AccessModifierOffset: -4
AlignAfterOpenBracket: Align
AlignConsecutiveAssignments: false
AlignConsecutiveDeclarations: false
AlignOperands: true
AlignTrailingComments: false
AllowAllParametersOfDeclarationOnNextLine: false
AllowShortBlocksOnASingleLine: false
AllowShortCaseLabelsOnASingleLine: false
AllowShortFunctionsOnASingleLine: None
AllowShortIfStatementsOnASingleLine: false
AllowShortLoopsOnASingleLine: false
AlwaysBreakAfterDefinitionReturnType: None
AlwaysBreakAfterReturnType: None
AlwaysBreakBeforeMultilineStrings: false
AlwaysBreakTemplateDeclarations: false
BinPackArguments: true
BinPackParameters: true

BraceWrapping:
  AfterClass: false
  AfterControlStatement: false
  AfterEnum: false
  AfterFunction: true
  AfterNamespace: true
  AfterObjCDeclaration: false
  AfterStruct: false
  AfterUnion: false
  AfterExternBlock: false # Unknown to clang-format-5.0
  BeforeCatch: false
  BeforeElse: false
  IndentBraces: false
  SplitEmptyFunction: true # Unknown to clang-format-4.0
  SplitEmptyRecord: true # Unknown to clang-format-4.0
  SplitEmptyNamespace: true # Unknown to clang-format-4.0

BreakBeforeBinaryOperators: None
BreakBeforeBraces: Custom
BreakBeforeInheritanceComma: false # Unknown to clang-format-4.0
BreakBeforeTernaryOperators: false
BreakConstructorInitializersBeforeComma: false
BreakConstructorInitializers: BeforeComma # Unknown to clang-format-4.0
BreakAfterJavaFieldAnnotations: false
BreakStringLiterals: false
ColumnLimit: 80
CommentPragmas: '^ IWYU pragma:'
CompactNamespaces: false # Unknown to clang-format-4.0
ConstructorInitializerAllOnOneLineOrOnePerLine: false
ConstructorInitializerIndentWidth: 4
ContinuationIndentWidth: 4
Cpp11BracedListStyle: false
DerivePointerAlignment: false
DisableFormat: false
ExperimentalAutoDetectBinPacking: false
FixNamespaceComments: false # Unknown to clang-format-4.0

IncludeBlocks: Preserve # Unknown to clang-format-5.0
IncludeCategories:
  - Regex: '.*'
    Priority: 1
IncludeIsMainRegex: '(Test)?$'
IndentCaseLabels: false
IndentPPDirectives: None # Unknown to clang-format-5.0
IndentWidth: 4
IndentWrappedFunctionNames: false
KeepEmptyLinesAtTheStartOfBlocks: false
MacroBlockBegin: ''
MacroBlockEnd: ''
MaxEmptyLinesToKeep: 1
NamespaceIndentation: None

PointerAlignment: Right
ReflowComments: false
SortIncludes: false
SortUsingDeclarations: false # Unknown to clang-format-4.0
SpaceAfterCStyleCast: false
SpaceAfterTemplateKeyword: true
SpaceBeforeAssignmentOperators: true
SpaceBeforeCtorInitializerColon: true # Unknown to clang-format-5.0
SpaceBeforeInheritanceColon: true # Unknown to clang-format-5.0
SpaceBeforeParens: ControlStatements
SpaceBeforeRangeBasedForLoopColon: true # Unknown to clang-format-5.0
SpaceInEmptyParentheses: false
SpacesBeforeTrailingComments: 1
SpacesInAngles: false
SpacesInContainerLiterals: false
SpacesInCStyleCastParentheses: false
SpacesInParentheses: false
SpacesInSquareBrackets: false
Standard: Cpp03
TabWidth: 4
UseTab: Never
```

jserv, 2021-09-02:
We can assume the version of `clang-format` >= 10 since [Ubuntu Linux 20.04-LTS](https://packages.ubuntu.com/focal/clang-format) is expected. Thus, the above can be cleaned up.

commit eef2bc43 (2021-09-02): Enforce Linux kernel coding style (#88)

The only exception is to indent with four spaces rather than tabs
for sticking to compact layout of source listing.

Close #87
closed 2021-09-02 by jserv