sysprog21/lkmpg #93: Several race conditions in chardev.c?

afiskon, 2021-09-05:
I believe there are several race conditions in chardev.c. The `msg` buffer is reused for several instances of the open file. If multiple processes will slowly read the same file this will cause memory corruption. Also, it seems like `open_device_cnt` should be atomic.

For simplicity, the best solution is probably simply to use a constant message. Also, I'm not an expert in Linux kernel modules. If there are in fact no bugs in the code, I think the text should explain in a little detail why this is the case.

afiskon, 2021-09-07:
@jserv  respectfully, the problem with:

```
static char msg[BUF_LEN];
```

... in chardev.c still exists. As I understand, `device_*` can be called concurrently which will cause concurrent access to `msg` and memory corruption as result. Unless there is something I don't know about how Linux kernel modules work. In the latest case I believe it should be clarified in the text and in the comments to the code.

jserv, 2021-09-07:
Reopen as per @afiskon requested. Under investigation.

YLowy, 2021-09-20:
I am testing the chardev.c on my machine and I think there is still some bug need to be fixed.

In my testing:
`[0] read 0 bytes.  
[1] read 0 bytes.  
[3] read 40 bytes. I already told you 1 times Hello world!
 
[2] read 40 bytes. I already told you 1 times Hello world!
 
[4] read 0 bytes.  
[3] read 40 bytes. I already told you 1 times Hello world!
 
[0] read 0 bytes.  
[1] read 40 bytes. I already told you 1 times Hello world!
 
[2] read 0 bytes.  
[4] read 40 bytes. I already told you 1 times Hello world!
`
Sometime, the read function from user space will fail (can't get the message) because the "Offset" in the open file table is not correct.
They all use the same open file table entry so there will also have race condition if we update the offset before device_read() return the bytes_read.

jserv, 2021-09-20:
> I am testing the chardev.c on my machine and I think there is still some bug need to be fixed.

@YLowy, you should discussion in the thread of pull request #108 if you were referring to the `offset` specific changes.

YLowy, 2021-09-21:
Due to the offset which is hard to be managed, I think it is a good way to create a list for each reading tasks. The element of the list include the task's PID and it's offset. So that the race condition in the open file table will not occur. Nevertheless, we still need to consider the list behavior when task close or lseek the device.

jserv, 2021-09-23:
Close via #108

commit 148fb013 (2021-09-07): Avoid unexpected concurrent access (#94)

In file {chardev,chardev2,sleep}.c, the variable to determine
the exclusive access was of integer type, which led to race
condition.

This patch rewrote the above with atomic CAS respectively
to eliminate the race.

Close #93
closed 2021-09-23 by jserv