sysprog21/lkmpg #99: CI: Use static analysis tools to verify the code

jserv, 2021-09-08:
Generally speaking, C programming language is not meant to be safety oriented, and we usually rely on extra static analysis tools to verify the code. For Linux kernel, [Sparse](https://www.kernel.org/doc/html/latest/dev-tools/sparse.html) is integrated into build systems. In addition, [Cppcheck](https://cppcheck.sourceforge.io/) is a neat tool to perform static analysis.

We can extend the existing CI pipeline to verify these examples before the built kernel modules are inserted into Linux.

Reference: [Assignment 3 - Software RAID](https://linux-kernel-labs.github.io/refs/heads/master/so2/assign3-software-raid.html)

jserv, 2021-09-15:
Close via #105 and #106

closed 2021-09-15 by jserv

fennecJ, 2021-09-23:
I checked the newest workflow, and it shows the following:
[image]

I am not familiar with sparse, shall we install `gtk2/gtk3` and `llvm` in docker or just ignore these 2 lines ?

jserv, 2021-09-23:
> I checked the newest workflow, and it shows the following:
> [image]
> 
> I am not familiar with sparse, shall we install `gtk2/gtk3` and `llvm` in docker or just ignore these 2 lines ?

`sparse` can be built without GTK and LLVM. Let's keep the minimal build for CI pipeline.