sysprog21/lkmpg #117: Make use of GCC Static Analysis

jserv, 2021-09-25:
Since gcc version 10, a new `-fanalyzer` option is implemented as a static analysis pass to identify various problems at compile-time, rather than at runtime.
See:
* [Static analysis in GCC 10](https://developers.redhat.com/blog/2020/03/26/static-analysis-in-gcc-10)
* [Options That Control Static Analysis](https://gcc.gnu.org/onlinedocs/gcc/Static-Analyzer-Options.html)
* [Improvements to GCC's -fanalyzer option](https://lwn.net/Articles/870290/)

We might make use of GCC static analysis to detect the potential problems in LKM. A possible way is to add the following into `Makefile`:
```
KBUILD_CFLAGS += \
    -Wall -Wundef \
    -Werror=strict-prototypes -Werror=implicit-function-declaration \
    -Werror=implicit-int \
    -fanalyzer -std=gnu89
```

linD026, 2021-09-25:
I tested, the example codes are no problem with `-fanalyzer` option.
But, when the module build with `gcc-10` it may get the error:
```shell
make -C /lib/modules/5.4.0-80-generic/build M=/home/ubuntu/lkmpg/examples modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-80-generic'
  CC [M]  /home/ubuntu/lkmpg/examples/hello-1.o
gcc-10: error: unrecognized command-line option ‘--param=allow-store-data-races=0’; did you mean ‘--allow-store-data-races’?
make[2]: *** [scripts/Makefile.build:271: /home/ubuntu/lkmpg/examples/hello-1.o] Error 1
make[1]: *** [Makefile:1762: /home/ubuntu/lkmpg/examples] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-80-generic'
make: *** [Makefile:38: all] Error 2
```
I made it to work by editing the build file, which is not a good choice in the **lkmpg**.
```shell
# Tell gcc to never replace conditional load with a non-conditional one
#KBUILD_CFLAGS  += $(call cc-option,--param=allow-store-data-races=0)
KBUILD_CFLAGS   += $(call cc-option,--allow-store-data-races=0)
KBUILD_CFLAGS   += $(call cc-option,-fno-allow-store-data-races)
```

---

```shell
diff --git a/examples/Makefile b/examples/Makefile
index 098f20b..ccd135e 100644
--- a/examples/Makefile
+++ b/examples/Makefile
@@ -32,6 +32,14 @@ obj-m += ioctl.o
 
 PWD := $(CURDIR)
 
+export CC=/usr/bin/gcc-10*
+
+KBUILD_CFLAGS += \
+    -Wall -Wundef \
+    -Werror=strict-prototypes -Werror=implicit-function-declaration \
+    -Werror=implicit-int \
+    -fanalyzer -std=gnu89
+
 all:
        $(MAKE) -C /lib/modules/$(shell uname -r)/build M=$(PWD) modules
```

linD026, 2021-10-05:
The `--param allow-store-data-races` internal parameter has been removed at GCC 10.
See [here](https://gcc.gnu.org/gcc-10/changes.html).
The patch added the `--param allow-store-data-races=0`, see [here](https://lore.kernel.org/lkml/alpine.LNX.2.00.1406101517300.9378@pobox.suse.cz/).

commit d6e1044a (2021-10-08): CI: introduce GCC static analysis

Since GCC version 10, there has been a new option -fanalyzer for
static analysis. It can make the CI pipeline more comprehensive.

Also, the static analysis updates in GCC 11, but we cannot install
the GCC 11 in ubuntu 20.04 straightforwardly right now, which is
the GitHub workflow environment (see status-check.yaml).
For this reason, we stay at the GCC 10.

Close #117

Reference
- https://gcc.gnu.org/onlinedocs/gcc-10.1.0/gcc/Static-Analyzer-Options.html
- https://developers.redhat.com/blog/2020/03/26/static-analysis-in-gcc-10
- https://lwn.net/Articles/8702 [... cut]