sysprog21/lkmpg #134: [Question] What is `insmod ’ing`?

mengxinayan, 2021-12-20:
I raise a question when I read section 5.2, What does `insmod ’ing` mean. I tried to search it in Google, but I didn't find anything:(. I'm not sure if this is a slip of a pen.

> 5.2 Functions available to modules
> Kernel modules are different here, too. In the hello world example, you might have noticed that we used a function, pr_info() but did not include a standard I/O library.  **That is because modules are object files whose symbols get resolved upon insmod ’ing**.  The definition for the symbols comes from the kernel itself; the only external functions you can use are the ones provided by the kernel. If you’re curious about what symbols have been exported by your kernel, take a look at /proc/kallsyms. 

Thanks.

hckuo, 2022-03-15:
It means installing a loadable kernel module via tools like `modprobe` or `insmod`.

jserv, 2022-03-16:
Thank @mengxinayan and @hckuo for clarifying.

commit 19521bc6 (2022-03-16): Refine wording on module loading

Close #134
closed 2022-03-16 by jserv