sysprog21/lkmpg #151: [Question]

QiuhaoLi, 2022-05-08:
In [13.2 Flashing keyboard LEDs](https://sysprog21.github.io/lkmpg/#flashing-keyboard-leds), it mentioned "Furthermore, the function prototype with unsigned long argument may be an obstacle to the control-flow integrity."

```c
struct timer_list { 
    unsigned long expires; 
    void (*function)(unsigned long);  // <---
    unsigned long data; 
    u32 flags; 
    /* ... */ 
}; 
```

**Could you make it clear why this argument prevents the control-flow integrity check?** As far as I know, hardware-assisted CFI like intel's [CET](https://www.intel.com/content/www/us/en/developer/articles/technical/technical-look-control-flow-enforcement-technology.html) uses shadow stack for ROP and IBT for JOP/COP, and there is nothing about parameters.

linD026, 2022-05-08:
Hi @QiuhaoLi ,
The CFI, like the Clang option [here](https://clang.llvm.org/docs/ControlFlowIntegrity.html#indirect-function-call-checking), will check the dynamic type of indirect function call to match the static type correctly. 
This kind of function prototype changes with CFI is for **forward-edge protection**.
The shadow stack that you mentioned is for **backward-edge protection**.

For more information about timer API changes, you can see [this](https://lwn.net/Articles/735887/).
Also, here is the [commit](https://github.com/torvalds/linux/commit/686fef928bba6be13cabe639f154af7d72b63120) which changes the API.

---
We didn't mention it because it might be too deep for the beginner.
We will consider again if this kind of description is compatible.
But still, thank you for the feedback!

QiuhaoLi, 2022-05-08:
Got it. Thanks for these helpful links! I think you can remove the paragraph about the API changes or give these links you mentioned (or maybe also [this one](https://nebelwelt.net/blog/20160913-ControlFlowIntegrity.html)) as additional read materials; otherwise, it's a little confusing for readers :)

PR #155 (2022-07-02): Add more information about the CFI of timer API
Currently, in 13.2 Flashing keyboard LEDs, the description of the
control-flow integrity, "Furthermore, the function prototype with
unsigned long argument may be an obstacle to the control-flow integrity
.", will confuse the reader.

It may lead the reader to think about hardware-assisted CFI like Intel
CET uses the shadow stack attack for ROP (backward-edge protection). But
the description of CFI here talks about the function pointer checking
with the prototype (forward-edge protection).

So add more information to make it more clear.

Close #15 [... cut]
commit 3d1fb290 (2022-07-02): Add more information about the CFI of timer API

Currently, in 13.2 Flashing keyboard LEDs, the description of the
control-flow integrity, "Furthermore, the function prototype with
unsigned long argument may be an obstacle to the control-flow integrity
.", will confuse the reader.

It may lead the reader to think about hardware-assisted CFI like Intel
CET uses the shadow stack attack for ROP (backward-edge protection). But
the description of CFI here talks about the function pointer checking
with the prototype (forward-edge protection).

So add more information to make it more clear.

Close #1 [... cut]
closed 2022-07-02 by jserv