sysprog21/lkmpg #158: Refine doc & code

minghu6, 2022-08-27:
1. Move the confusing note about `struct proc_ops` replacing with `struct file_operations` on linux >=5.6:
https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/lkmpg.tex#L957-L990
The `struct proc_ops` would replace the `struct file-operations` only when create proc  releated defined on <linux/proc_fs.h>
In other word, regist device doesn't be affected, still using `struct file_operations`. It's confusing write this note above the regist device code (FOE ME, I HAVE SPENT PLENTY OF TIME ON SEARCHING HOW TO REGIST DEVICE USING STRUCT PROC_OPS)
I highly recommend to move this note down to next code example which replacing takes place truely!

2.  https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/examples/chardev.c#L25-L39
The msg char array should be declared with length of `BUF_LEN + 1` since `BUF_LEN` is max length of message.

minghu6, 2022-09-02:
5. Unmached code and comment
https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/examples/print_string.c#L44-L#56
`"\015\012"` is `\v\b` rather than CRLF `\r\n`.

linD026, 2022-09-07:
Hi @minghu6 , 
Sorry for the late to reply.

> 1. Move the confusing note about `struct proc_ops` replacing with `struct file_operations` on linux >=5.6: [...]

Yes, it won't affect the registering of the device, but the paragraph here actually talks about the file operation. And, it seems like the file operation will be related to the proc_ops. So, I think the description of proc_ops replacing the file_ops is still needed in the section here.
But still, it might confuse others. So, I will add more sentences to hint that this description talks about the file_ops and proc_ops rather than registering the device.

> 2. https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/examples/chardev.c#L25-L39
>    
>    The msg char array should be declared with length of `BUF_LEN + 1` since `BUF_LEN` is max length of message.

The macro, `BUF_LEN`, is the length of the buffer, not the message length.
So, I think there doesn't need to change.

> 3. https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/examples/chardev.c#L152
>    
>    On `device_write`, It's prefered to use `EOPNOTSUPP` instead of `EINVAL` since the former raises unsupported error, the latter raises invalid argument.

Yes, it should uses the `EOPNOTSUPP`.
Thanks.

> 4. Improve read/write fucntion code among `fsproc{1-3}.c`:
>    https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/examples/procfs3.c#L23-L40
>    
>    https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/examples/procfs3.c#L41-L53
> 
> These code are a little wired and far from decent, the parameter `loff_t *` is ignored incorrectly and so the behavior is quite different from normal read/write.I've reimplemented read/write code, here is full code:
> 
> It need more modern c standard config than C90, for example `ccflags-y += -std=gnu17`.

The read/write functions are a little weird. I will modify them.
For using modern C standards, we need to consider more.
Since kernel version v5.18, especially after this commit ["Kbuild: move to -std=gnu11"](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/Makefile?h=v6.0-rc4&id=e8c07082a810fbb9db303a2b66b66b8d7e588b53), kernel supports C11 features. We have the option to move to it.
But considering LKMPG is supporting the v5.x version, we must stick to the C89 still a while.
However, for C17, it might be the wrong standard to use.

linD026, 2022-09-07:
> 5. Unmached code and comment
>    https://github.com/sysprog21/lkmpg/blob/637e707a1ab7efa62234a374c62ffb040800aca8/examples/print_string.c#L44-L56
>    
>    `"\015\012"` is `\v\b` rather than CRLF `\r\n`.

The `"\015\012"` seems correct.
According to [the ASCII table](https://en.cppreference.com/w/cpp/language/ascii), `CR` is `015` and `LF` is `012`.
And, `\v\b` is `\013\010`.
I might be wrong, please provide more information about this.

Thanks

linD026, 2022-09-07:
> Yes, it should uses the `EOPNOTSUPP`. Thanks.

Sorry, I recheck the write man page to make sure that the change here is suitable to the definition.
In the man page, it doesn't have any error value that is `EOPNOTSUPP`, `NOTSUPP`, or even `NOTSUPP` (kernel defined).
And from the description of `EINVAL` in the manual:

```
EINVAL **fd is attached to an object which is unsuitable for
       writing**; or the file was opened with the O_DIRECT flag,
       and either the address specified in buf, the value
       specified in count, or the file offset is not suitably
       aligned.
```

The `EINVAL` is correct.

linD026, 2022-09-07:
> The macro, `BUF_LEN`, is the length of the buffer, not the message length. So, I think there doesn't need to change.

Oops, I saw the comments.
Sorry again. But I still don't think that we should change here.
The method we measure the length of the message includes the '\0' which I think that it s why you suggest that it needs to add one more byte.

jserv, 2022-09-08:
Ready to close?

linD026, 2022-09-08:
Sure, I think.
If there are any other opinions we can just reopen it.

closed 2022-09-08 by linD026