sysprog21/lkmpg #165: Reopen the issue #160

minghu6, 2022-09-11:
En..., I have to reopen it for the fact as follows:
The field `f_op` used by `vfs_xx` is a wrapper of `struct file_operations` or `struct proc_operations`.

https://elixir.bootlin.com/linux/v5.15/source/fs/proc/inode.c#L644
https://elixir.bootlin.com/linux/v5.15/source/fs/proc/inode.c#L570
https://elixir.bootlin.com/linux/v5.15/source/fs/proc/inode.c#L276

**And the warpper always be nonnull even if the raw op function is null** (I think this weakness shouldbe treated as kind of bug for Linux Kernel)

The crash log has verified it:
```
[419136.210798] /proc/hello created
[419136.210807] /sys/kernel/hi: Init.
[419136.211108] Flywheel ready
[419136.409492] Turn the crank
[419136.409531] Flywheel spins up
[419136.425516] SymmetricKey: password123
[419136.425519] PlainText: THIS IS A PLAIN TEXT
[419159.628887] BUG: kernel NULL pointer dereference, address: 0000000000000000
[419159.628892] #PF: supervisor instruction fetch in kernel mode
[419159.628893] #PF: error_code(0x0010) - not-present page
[419159.628894] PGD 0 P4D 0 
[419159.628897] Oops: 0010 [#1] PREEMPT SMP PTI
[419159.628900] CPU: 2 PID: 419062 Comm: tail Kdump: loaded Tainted: P           OE     5.16.0+ #3
[419159.628902] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./B85M Pro3, BIOS C1.20 07/03/2014
[419159.628903] RIP: 0010:0x0
[419159.628908] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[419159.628909] RSP: 0018:ffffc90003a43ec0 EFLAGS: 00010246
[419159.628911] RAX: 0000000000000000 RBX: ffff88803d839cc0 RCX: 0000000000000001
[419159.628912] RDX: 0000000000000001 RSI: 0000000000000000 RDI: ffff88830a589a00
[419159.628913] RBP: ffffc90003a43ed8 R08: 0000000000000008 R09: 0000000000000000
[419159.628914] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000001
[419159.628915] R13: 0000000000000000 R14: ffffffffffffffea R15: ffff88830a589a00
[419159.628916] FS:  00007f965833e740(0000) GS:ffff88840f700000(0000) knlGS:0000000000000000
[419159.628918] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[419159.628919] CR2: ffffffffffffffd6 CR3: 00000002e60d4005 CR4: 00000000001706e0
[419159.628921] Call Trace:
[419159.628923]  <TASK>
[419159.628924]  proc_reg_llseek+0x4e/0x80
[419159.628928]  ? __fdget_pos+0x17/0x50
[419159.628933]  ksys_lseek+0x84/0xc0
[419159.628937]  __x64_sys_lseek+0x18/0x20
[419159.628940]  do_syscall_64+0x5c/0xc0
[419159.628943]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[419159.628945] RIP: 0033:0x7f9658455acb
[419159.628947] Code: ff ff c3 0f 1f 40 00 48 8b 15 61 43 10 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb ba 0f 1f 00 f3 0f 1e fa b8 08 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 05 c3 0f 1f 40 00 48 8b 15 31 43 10 00 f7 d8
[419159.628948] RSP: 002b:00007ffe702e7fd8 EFLAGS: 00000246 ORIG_RAX: 0000000000000008
[419159.628950] RAX: ffffffffffffffda RBX: 0000563b374544a0 RCX: 00007f9658455acb
[419159.628952] RDX: 0000000000000001 RSI: 0000000000000000 RDI: 0000000000000003
[419159.628953] RBP: 00007ffe702e888b R08: 0000000000000001 R09: 0000563b37454440
[419159.628954] R10: 0000000000001000 R11: 0000000000000246 R12: 00007ffe702e888b
[419159.628955] R13: 0000000000000000 R14: 0000563b363123f9 R15: 0000000000000003
[419159.628957]  </TASK>

```

minghu6, 2022-09-11:
Sincerely, My system did has crashed, and supply an implement did fixed it!

linD026, 2022-09-13:
Interesting. I will deal with it.
Thanks for the information!

PR #168 (2022-09-23): Fix dereference NULL pointer with proc_lseek
Since the operations are static storage duration, the pointer in operation structures will initialize with NULL. But, the kernel doesn't check whether the pointer is NULL or not when calling it.

Related Discussion:
- https://github.com/sysprog21/lkmpg/issues/165
- https://github.com/sysprog21/lkmpg/issues/160

Close #165
commit 0f9c7263 (2022-09-23): Fix dereference NULL pointer with proc_lseek

Since the operations are static storage duration, the pointer in
operation structures will initialize with NULL. But, the kernel
doesn't check whether the pointer is NULL or not when calling it.

Related Discussion:
- https://github.com/sysprog21/lkmpg/issues/165
- https://github.com/sysprog21/lkmpg/issues/160

Close #165
closed 2022-09-23 by jserv