sysprog21/lkmpg #172: Support Linux v6.0

jserv, 2022-10-08:
At present, Linux v5.10+ is the primary kernel version for LKMPG. Meanwhile, Linux v6.0 was released, and it is time to check the availability of existing LKM with Linux v6.0.

linD026, 2022-10-08:
Just take note and request some opinions here.
If we want to add the Rust Kernel Module, what kind of thing do we need to prepare?
My first mind is we can follow these tutorials for the environment setup.
- https://youtu.be/-l-8WrGHEGI
- https://youtu.be/tPs1uRqOnlk

Then, add some other example that is not the same as the existing one.
Or we can just rewrite the examples by the Rust.

jserv, 2023-02-08:
[Linux 6.1 is officially promoted to be an LTS kernel](https://www.phoronix.com/news/Linux-6.1-LTS-Official), and we shall re-target v6.1 compatibility.

ashevche, 2023-02-21:
Agree, the book should cover a single or a few LTS kernels and not just something in the middle.

linD026, 2023-02-22:
Sure, it makes sense to add the version we covered.
However, in `1.1 Authorship`, we already told it.
>  Jim Huang upgraded to recent kernel versions (v5.x) and revised the LaTeX document.

linD026, 2023-03-14:
Just take some notes here.
For 6.1.11 and gcc (GCC) 12.2.1 20230201, I got the following compile error.
```
/home/***/lkmpg/examples/kbleds.c:64:55: error: ‘struct tty_driver’ has no member named ‘magic’
   64 |     pr_info("kbleds: tty driver magic %x\n", my_driver->magic);
      | 
```

jserv, 2023-03-14:
> Just take some notes here. For 6.1.11 and gcc (GCC) 12.2.1 20230201, I got the following compile error.

Once you resolve the compilation error, let's bump the gcc versions in GitHub Actions.

jserv, 2023-09-17:
> Just take some notes here. For 6.1.11 and gcc (GCC) 12.2.1 20230201, I got the following compile error.
> ```
> examples/kbleds.c:64:55: error: ‘struct tty_driver’ has no member named ‘magic’
>    64 |     pr_info("kbleds: tty driver magic %x\n", my_driver->magic);
>       | 
> ```

Once you resolve the compilation error, let's bump the gcc versions in GitHub Actions.

jserv, 2024-10-26:
Recent CI pipelines already trigger Linux v6.x builds.

closed 2024-10-26 by jserv