sysprog21/lkmpg #188: lkmpg/Makefile all

tclee060219sat, 2023-03-13:
My laptop is running 5.19.0-35-generic on it. My gcc version is gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0. 

I am resolving the "lkmpg/Makefile all" issues shown below. Comment please.

tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$ make all
.....
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/completions.c: In function ‘machine_crank_thread’:
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/completions.c:21:5: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
   21 |     complete_and_exit(&machine.crank_comp, 0);
      |     ^~~~~~~~~~~~~~~~~
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/completions.c:22:1: error: no return statement in function returning non-void [-Werror=return-type]
   22 | }
      | ^
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/completions.c: In function ‘machine_flywheel_spinup_thread’:
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/completions.c:32:1: error: no return statement in function returning non-void [-Werror=return-type]
   32 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:257：/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/completions.o] 错误 1
make[1]: *** [Makefile:1850：/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples] 错误 2
make[1]: 离开目录“/usr/src/linux-headers-5.19.0-35-generic”
make: *** [Makefile:52：all] 错误 2
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$

tclee060219sat, 2023-03-13:
I commented out "# mar13mon23 obj-m += completions.o. The .../lkmpg/examples$ make clean&&make" execution completed with no issue. Will suspend the the completions.c building errors upon request. Comment please.

linD026, 2023-03-14:
Hi, It seems like, after v5.17-rc1, complete_and_exit is changed to kthread_complete_and_exit.
I will fix this ASAP.
Thanks.

tclee060219sat, 2023-03-14:
Thanks for your response. Once you fix it I should do the GIT command to update my local lkmpg repo.

linD026, 2023-03-14:
Sure.
Once the [PR](https://github.com/sysprog21/lkmpg/pull/189) being merged, you can fetch the branch that already fixed the errors and rebase to it.

commit 4ee80a3e (2023-03-14): completion: Improve the compatibility with v5.17+

Since v5.17-rc1, particularly after the commit cead1855266 ("exit:
Rename complete_and_exit to kthread_complete_and_exit"),
complete_and_exit() is renamed to kthread_complete_and_exit().

Close #188
closed 2023-03-14 by jserv