sysprog21/lkmpg #190: KernelModuleLUl.sh

tclee060219sat, 2023-03-14:
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$ ../../../BASH/KernelModuleLUl.sh hello
../../../BASH/KernelModuleLUl.sh@2
../../../BASH/KernelModuleLUl.sh@3: hello
../../../BASH/KernelModuleLUl.sh@4: 1
rmmod: ERROR: Module hello_1 is not currently loaded
-----@13 unloading failure
-----@15 loading success
hello_1                16384  0
-----@18 unloading success
3月 14 15:13:38 tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA kernel: Hello world 1.
3月 14 15:13:38 tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA kernel: Goodbye world 1.
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$

Shown above is the result of my BASH script which loads and unloads hello-1.ko. Listed below is the script for review. The reason for developing this suite of scripts is for exercising the Linux kernel module work efficiently. Comment please:

#!/usr/bin/bash
sudo echo $0@$LINENO # command name 
sudo echo $0@$LINENO: $@ # whole arguments 
sudo echo $0@$LINENO: $# #  number of arguments 
if [[ "$#" != 1 ]]; then
    echo "Usage: $0 <kernelmodule>"
    exit
fi
#if sudo rmmod hello_1; then
if sudo rmmod $1_1; then
    echo -----@$LINENO unloading success
else
    echo -----@$LINENO unloading failure 
    if sudo insmod $1-1.ko; then
        echo -----@$LINENO loading success
        sudo lsmod|egrep hello
        if sudo rmmod $1_1; then
            echo -----@$LINENO unloading success
                sudo journalctl --since "0.3 min ago" | grep kernel|egrep -v UFW
        else
            echo -----@$LINENO unloading failure 
        fi
    else
        echo -----@$LINENO loading failure
        exit
    fi
fi

tclee060219sat, 2023-03-15:
A new version of KernelModuleLUl.sh, listed below, can take care of hello-1/2/3/4//5. Comment please.

#!/usr/bin/bash
sudo echo $0@$LINENO # command name 
sudo echo $0@$LINENO: $@ # whole arguments 
sudo echo $0@$LINENO: $# #  number of arguments 
if [[ "$#" != 2 ]]; then
    echo "Usage: $0 <kernelmodule> <index>"
    exit
fi
if sudo rmmod $1_$2; then
    echo -----@$LINENO unloading success
else
    echo -----@$LINENO unloading failure 
    if [[ "$2" == 5 ]]; then
        echo ----@$LINENO Read in two auguments
        read mystring
            echo ---@$LINENO mystring == $mystring
        read myintarray
            echo ---@$LINENO myintarray == $myintarry
            sudo insmod $1-$2.ko
            sudo insmod $1_$2.ko mystring=$mystring myintarray= $myinsarray
    else
        sudo insmod $1-$2.ko
    fi
    echo -----@$LINENO loading success
        sudo lsmod|egrep hello
    if sudo rmmod $1_$2; then
        echo -----@$LINENO unloading success
        sudo journalctl --since "0.3 min ago" | grep kernel|egrep -v UFW
    else
        echo -----@$LINENO unloading failure 
    fi
fi

jserv, 2023-03-15:
Can you please send a pull request?

linD026, 2023-03-15:
Please use a code block to display your program or script, and make sure to indent it properly.
For the script, please provide an example command, for example:

```
bash KernelModuleLUl.sh hello 5
```

Also, make sure that the file name of the script follows the Linux naming style.
You can find more information [here](https://www.kernel.org/doc/html/v4.10/process/coding-style.html#naming).
Although this convention is in the C language, but it is still relevant for the file name.

tclee060219sat, 2023-03-15:
To jserv, as a new team member I need to figure out how to do "pull request". You are welcome to give me the details of this request.

jserv, 2023-03-15:
> To jserv, as a new team member I need to figure out how to do "pull request". You are welcome to give me the details of this request.

Please check this: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

tclee060219sat, 2023-03-15:
To linD026, I will refer to https://www.kernel.org/doc/html/v4.10/process/coding-style.html#naming to follow the Linux naming style.
As a new team member, I need  to know how to use code block to display my script. You are welcome to give the details of "code block".

jserv, 2023-03-15:
>  I need to know how to use code block to display my script. You are welcome to give the details of "code block".

Check this: https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

tclee060219sat, 2023-03-16:
I did a call request and archived scripts/kernel-module-exercise.sh. Shown below is result of my local repo archiving execution. For this work I used Linux Naming Style.  I will use code-block in the future if it becomes necessary. My next step is to archive this script into the lkmpg remote repo. Comment please. 

tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg$ git add &&git commit =m "Initial.version" scripts/kernel-module-exercise.sh&&git log --stat scripts/kernel-module-exercise.sh
error: 路径规格 '=m' 未匹配任何 git 已知文件
error: 路径规格 'Initial.version' 未匹配任何 git 已知文件
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg$ git add scripts/kernel-module-exercise.sh&&git commit -m "Initial.version" scripts/kernel-module-exercise.sh&&git log --stat scripts/kernel-module-exercise.sh
[master 17744bd] Initial.version
 1 file changed, 42 insertions(+)
 create mode 100755 scripts/kernel-module-exercise.sh
commit 17744bd6f91f17dad9a768d7562c6abfa520290b (HEAD -> master)
Author: tclee0209thu23 <tommyclee101908sun@gmail.com>
Date:   Thu Mar 16 15:16:35 2023 -0700

    Initial.version

 scripts/kernel-module-exercise.sh | 42 ++++++++++++++++++++++++++++++++++++++++++
 1 file changed, 42 insertions(+)
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg$

tclee060219sat, 2023-03-16:
I will not archive my work into the remote repo until the receipt of your comment on my local repo work from you.

linD026, 2023-03-18:
I'm sorry for not getting back to you sooner. I have been unwell recently.

I think you misunderstood something here.
When sharing sample code or terminal output on GitHub's comment section (e.g., issue) using a code block can help readers better understand the code.
For the result you shared, it is just a series of git commands, I cannot provide any comment on it.

I suggest that you can send the pull request and let us review the script and the content of the git commit rather than discussing it here.

And please make sure the script is runnable. I saw the error in the first comment:

```
rmmod: ERROR: Module hello_1 is not currently loaded
```

I think the problem is the module name is `hello-*` instead of `hello_*`.

Thanks.

tclee060219sat, 2023-03-20:
Last week I did a call request and archived scripts/kernel-module-exercise.sh locally. Please tell me how to do the sharing of my script and its commands sequence into the lkmpg remote repo, so that the team members can review the script.  

Please notice the change of my script name, per Linux naming Style.
At the moment my script is still in the development to cover the exercise, i.e. module load/unload, of all the .ko files.

linD026, 2023-03-22:
You can see [this](https://docs.github.com/en/get-started/using-git/pushing-commits-to-a-remote-repository).
And check [this out](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request).

tclee060219sat, 2023-03-22:
I will do these two steps when I need to. Thanks for your comment.

jserv, 2023-03-24:
Let's break down the discussions in more specific manner.

closed 2023-03-24 by jserv