sysprog21/lkmpg #192: procfs1.ko runtime investigation

tclee060219sat, 2023-03-23:
Shown below is my procfs1.ko runtime investigation.  It shows "cat /proc/helloworld" runs only once but procfile_read() in procfs1.c is called twice. I am interested in knowing why procfile_read() is called twice for one run of "cat /proc/helloworld". Comment please.

Terminal 1 (/dev/pts/1)
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$ make&&../scripts/kernel-module-exercise.sh procfs1
Makefile:37: $(CURDIR) == /media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples
Makefile:38: $(CONFIG_STATUS_CHECK_GCC) == 
make -C /lib/modules/5.19.0-35-generic/build CC=cc M=/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples modules
make[1]: 进入目录“/usr/src/linux-headers-5.19.0-35-generic”
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  You are using:           cc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/Makefile:37: $(CURDIR) == /usr/src/linux-headers-5.19.0-35-generic
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/Makefile:38: $(CONFIG_STATUS_CHECK_GCC) == 
  CC [M]  /media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/procfs1.o
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/Makefile:37: $(CURDIR) == /usr/src/linux-headers-5.19.0-35-generic
/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/Makefile:38: $(CONFIG_STATUS_CHECK_GCC) == 
  MODPOST /media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/Module.symvers
  CC [M]  /media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/procfs1.mod.o
  LD [M]  /media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/procfs1.ko
  BTF [M] /media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/procfs1.ko
Skipping BTF generation for /media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples/procfs1.ko due to unavailability of vmlinux
make[1]: 离开目录“/usr/src/linux-headers-5.19.0-35-generic”
-----@34 $mnrm == procfs1
rmmod: ERROR: Module procfs1 is not currently loaded
-----@21 module_remove() procfs1 unloading failure
-----@46 $pstr ==
-----@48 loading success
Terminal 2 (/dev/pts/4)
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/chardev$ cat /proc/helloworld 
HelloWorld!
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/chardev$ 
Terminal 1 c for continue, else for quit: c
-----@19 module_remove() procfs1 unloading success
3月 23 11:04:38 tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA kernel: /proc/helloworld created
3月 23 11:04:47 tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA kernel: procfile read helloworld
3月 23 11:04:47 tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA kernel: copy_to_user failed
3月 23 11:04:57 tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA kernel: /proc/helloworld removed
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$

jserv, 2023-03-24:
The messages above have been condensed. I defer to @linD026 .

linD026, 2023-03-24:
This question is already answered in the  `/proc File System` chapter:

> Every time the file /proc/helloworld is read, the function procfile_read is called. Two parameters of this function are very important: the buffer (the second parameter) and the offset (the fourth one). The content of the buffer will be returned to the application which read it (for example the cat command). The offset is the current position in the file. If the return value of the function is not null, then this function is called again. So be careful with this function, if it never returns zero, the read function is called endlessly.

So, the first time it will read the buffer which is "HelloWorld!", then add `len` to `offset` and return a non-zero value to indicate the file system calls the function again. The second time, the condition check of `offset` will let the function return a zero value so it won't be called again.
This is why `procfile_read()` is called twice.

closed 2023-03-24 by linD026