sysprog21/lkmpg #194: The "sudo lsmod|egrep procfs3" shows -1 on the "used by" column

tclee060219sat, 2023-03-30:
Shown below is the procfs3 insmod/lsmod/rmmod scenario. I am figuring out why -1 is on its 3rd column. Comment please. FYI, the same command sequence for procfs1.ko and procfs2.ko succeeded. 

tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$ ls -lrt procfs3.*
-rw-rw-r-- 1 tclee tclee     98  3月 27 14:37 procfs3.mod
-rw-rw-r-- 1 tclee tclee     98  3月 27 14:37 procfs3.mod
-rw-rw-r-- 1 tclee tclee   3445  3月 30 14:54 procfs3.c.mar30thu23.0254pm
-rw-rw-r-- 1 tclee tclee   1253  3月 30 15:24 procfs3.mod.c
-rw-rw-r-- 1 tclee tclee  50200  3月 30 15:24 procfs3.mod.o
-rw-rw-r-- 1 tclee tclee   3021  3月 30 15:34 procfs3.c
-rw-rw-r-- 1 tclee tclee 166288  3月 30 15:34 procfs3.o
-rw-rw-r-- 1 tclee tclee 214984  3月 30 15:34 procfs3.ko
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$ sudo lsmod|egrep procfs
procfs3                16384  -1
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$ sudo rmmod procfs3
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$ sudo lsmod|egrep procfs
procfs3                16384  -1
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C/lkmpg/examples$

jserv, 2023-03-31:
@tclee060219sat, I annotated your message for better illustration. Please do that always.

linD026, 2023-03-31:
Not sure what is happening on your side, although it might be related to the `try_module_get` [[1](https://github.com/sysprog21/lkmpg/issues/52)].
I don't have this problem with 5.4.0-80-generic kernel:
```bash
ubuntu@ubuntu:~/lkmpg/examples$ sudo insmod procfs3.ko
ubuntu@ubuntu:~/lkmpg/examples$ sudo lsmod|egrep procfs3
procfs3                20480  0
ubuntu@ubuntu:~/lkmpg/examples$ sudo rmmod procfs3 
ubuntu@ubuntu:~/lkmpg/examples$ sudo lsmod|egrep procfs3
ubuntu@ubuntu:~/lkmpg/examples$ 
```
Could you please provide more information?

closed 2023-03-31 by jserv