sysprog21/lkmpg #195: procfs3.ko used by -1

tclee060219sat, 2023-04-04:
The "sudo insmod procfs3.ko" command hangs the Ubuntu 5.19.0-38-generic laptop. In another terminal, the "lsmod|egrep procfs3" execution shows:

tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C$ lsmod|egrep procfs3
procfs3                16384  -1
tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:/media/tclee/1081336d-1324-4fc9-acb6-2b05d12662b0/HomeWork/Programming/C$ sudo rmmod procfs3

I discovered the solution to this problem is the power-cycle of this laptop. I am seeking the more efficient solution.

Comment please.

linD026, 2023-04-27:
It seems like when the `lsmod` calls to the `procfs_open` and `procfs_close`, it doesn't increase but decreases the counter.
Currently, I don't have the v5.19 environment, so I cannot figure out why this happened.

tclee060219sat, 2023-04-28:
A module reference count of -1, visible both in /sys/module/<module>/refcnt and in lsmod’s output, means that the module is currently unloading.

If a module’s reference count stays at -1, that indicates a problem — dmesg should tell you more.

jserv, 2024-09-26:
Can not reproduced.

closed 2024-09-26 by jserv

tclee060219sat, 2024-09-26:
I am now working on two platforms:
DellLatitude-E6230:Ubuntu 24.04.1 LTS 64bits
Rpi3MB:Raspbian GNU/Linux 12 (bookworm) 32bits

The loading/unloading of procfs3.ko on them has no problems at all. For the issue happened while running on Ubuntu 5.19.0-38-generic laptop, I will not bother to resolve it.

My current work is the enabling of debugging LKMPG modules using gdb on DellLatitude-E6230 (host) and kdgb on Rpi3MB (tartget). The connection between them is minicom over FT232R (no issues).