sysprog21/lkmpg #199: Reason to reset the offset

franzwong, 2023-04-30:
Sorry if it is not the place to ask question.

On line 127 of chardev.c, when it reaches the end of message, why do we need to reset the offset back to zero?

Snippet:

```
static ssize_t device_read(struct file *filp, /* see include/linux/fs.h   */ 
                           char __user *buffer, /* buffer to fill with data */ 
                           size_t length, /* length of the buffer     */ 
                           loff_t *offset) 
{ 
    /* Number of bytes actually written to the buffer */ 
    int bytes_read = 0; 
    const char *msg_ptr = msg; 
 
    if (!*(msg_ptr + *offset)) { /* we are at the end of message */ 
        *offset = 0; /* reset the offset */ 
        return 0; /* signify end of file */ 
    } 
```

linD026, 2023-04-30:
Hi,
It's just making sure that every time we read to the end of the buffer we go back to the beginning.
Even though, in most cases, we just open (create the empty file) the file, read it, and close it once time, the reset might be unnecessary.
When we open the file, the function calls in v5.4 will be like this:

```
[113739.851760]  device_open.cold+0x18/0x4b [chardev]
[113739.851762]  chrdev_open+0xd3/0x1c0
[113739.851764]  ? cdev_default_release+0x20/0x20
[113739.851765]  do_dentry_open+0x143/0x3a0
[113739.851767]  vfs_open+0x2d/0x30
[113739.851769]  do_last+0x194/0x900
[113739.851770]  path_openat+0x8d/0x290
[113739.851771]  do_filp_open+0x91/0x100
[113739.851774]  ? __alloc_fd+0x46/0x150
[113739.851775]  do_sys_open+0x17e/0x290
[113739.851777]  __x64_sys_openat+0x20/0x30
[113739.851779]  do_syscall_64+0x57/0x190
[113739.851780]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
```
See the [path_openat()](https://elixir.bootlin.com/linux/v5.4.242/source/fs/namei.c#L3513), it will allocate the empty file:

```c
static struct file *path_openat(struct nameidata *nd,
			const struct open_flags *op, unsigned flags)
{
	struct file *file;
	int error;

	file = alloc_empty_file(op->open_flag, current_cred());
	if (IS_ERR(file))
		return file;
```
So, every time we do the open-read-close, the offset will be reset implicitly.
However, if we do something like open-read-read-close, we need to reset the offset by ourselves, otherwise, we will not get any data on the subsequent read calls.

Hopefully, this will help you.
Thanks :)

franzwong, 2023-05-01:
I understand now. Thanks for the detailed explanation.

closed 2023-05-01 by franzwong