sysprog21/lkmpg #204: Building kernel modules in RpiOS32bit/Rpi3MB platform

tclee060219sat, 2023-06-07:
Eventually I made the kernel header and kernel source matched on my RpiOS32bit/Rpi3MB platform. Now I am moving onto the exercise of the kernel modules I am interested on this platform.

tclee060219sat, 2023-06-07:
tclee@raspberrypi:~/jun07wed23/rpilkm/RPI-LKM $ ls -l /usr/src/
total 4
drwxr-xr-x 3 root root 4096 May 2 17:14 sense-hat
tclee@raspberrypi:~/jun07wed23/rpilkm/RPI-LKM $ sudo apt install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
libfuse2
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
raspberrypi-kernel-headers
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 30.0 MB of archives.
After this operation, 193 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian bullseye/main armhf raspberrypi-kernel-headers armhf 1:1.20230405-1 [30.0 MB]
Fetched 30.0 MB in 3min 4s (163 kB/s)
apt-listchanges: Can't set locale; make sure $LC_* and $LANG are correct!
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
LANGUAGE = (unset),
LC_ALL = (unset),
LC_TIME = "zh_CN.UTF-8",
LC_MONETARY = "zh_CN.UTF-8",
LC_ADDRESS = "zh_CN.UTF-8",
LC_TELEPHONE = "zh_CN.UTF-8",
LC_NAME = "zh_CN.UTF-8",
LC_MEASUREMENT = "zh_CN.UTF-8",
LC_IDENTIFICATION = "zh_CN.UTF-8",
LANG = "en_GB.UTF-8"
are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_ALL to default locale: No such file or directory
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 106933 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1%3a1.20230405-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1:1.20230405-1) ...
Setting up raspberrypi-kernel-headers (1:1.20230405-1) ...
tclee@raspberrypi:~/jun07wed23/rpilkm/RPI-LKM $ ls -l /usr/src/
total 16
drwxr-xr-x 25 root root 4096 Jun 7 11:56 linux-headers-6.1.21+
drwxr-xr-x 25 root root 4096 Jun 7 11:56 linux-headers-6.1.21-v7+
drwxr-xr-x 25 root root 4096 Jun 7 11:56 linux-headers-6.1.21-v7l+
drwxr-xr-x 3 root root 4096 May 2 17:14 sense-hat
tclee@raspberrypi:~/jun07wed23/rpilkm/RPI-LKM $ ls /lib/modules/$uname -r)
-bash: syntax error near unexpected token `)'
tclee@raspberrypi:~/jun07wed23/rpilkm/RPI-LKM $ ls /lib/modules/$(uname -r)
build modules.alias modules.builtin modules.builtin.bin modules.dep modules.devname modules.softdep modules.symbols.bin
odules.dep.bin modules.order modules.symbols
tclee@raspberrypi:~/jun07wed23/rpilkm/RPI-LKM $

jserv, 2023-06-08:
> Eventually I made the kernel header and kernel source matched on my RpiOS32bit/Rpi3MB platform. Now I am moving onto the exercise of the kernel modules I am interested on this platform.

Please report issues if you found something wrong or worthy for improvements. Otherwise, I am about to close this.

closed 2023-06-08 by jserv