sysprog21/lkmpg #217: Building kbleds.ko on RpiOSLite64bit (Linux 6.1.21-v8+)

tclee060219sat, 2023-08-09:
Shown below is the building error. On Ubuntu22.04.2LTS there are no building errors at all. I will resolve this error if it is necessary.

tclee@raspberrypi:~/aug09wed23/lkmpg/examples $ make clean&&make&&ls -lrt *.ko
...../home/tclee/aug09wed23/lkmpg/examples/kbleds.c: In function ‘kbleds_init’:
/home/tclee/aug09wed23/lkmpg/examples/kbleds.c:64:55: error: ‘struct tty_driver’ has no member named ‘magic’
   64 |     pr_info("kbleds: tty driver magic %x\n", my_driver->magic);

tclee060219sat, 2023-08-09:
My git clone command is:

$ git clone https://github.com/sysprog21/lkmpg.git && cd lkmpg

tclee060219sat, 2023-08-09:
Discovery shown as below:
Ubuntu22.04.2LTS:
struct tty_driver {tclee@tclee-VivoBook-12-ASUS-Laptop-E203MAS-E203MA:~$ vim /usr/src/linux-hwe-5.19-headers-5.19.0-50/include/linux/tty_driver.h
struct tty_driver {
    int magic;
    struct kref kref;
    .....
}
tclee@raspberrypi:~/aug09wed23/lkmpg/examples $ vim /lib/modules/6.1.21-v8+/build/include/linux/tty_driver.h
struct tty_driver {
    struct kref kref;
   .....
};

jserv, 2023-08-09:
Recent commit (https://github.com/torvalds/linux/commit/5052df99d3bc3cd281222bbcba44323b2d0937d2) removes `TTY_DRIVER_MAGIC`:
> According to Greg, in the context of magic numbers as defined in magic-number.rst, "the tty layer should not need this and I'll gladly take patches"

@linD026, can you confirm and see if a quick fix is available to drop the reference to `magic` field?

linD026, 2023-08-10:
Sure, I saw this [mailing list](https://lore.kernel.org/linux-doc/20220907231109.htnfxzmr6wkuhn2v@tarta.nabijaczleweli.xyz/), it seems like we can safely remove the `magic` field.
I will send the patch ASAP.

closed 2023-08-10 by jserv

tclee060219sat, 2023-08-10:
kbleds.c has only one line of accessing magic variable:
67:    // pr_info("kbleds: tty driver magic %x\n", my_driver->magic);
Commented this line out, no more building error in rpiOSLite64bit.

PR #218 (2023-08-10): kbleds: Replace magic with driver_name
Since the commit 5052df99d3bc ("tty: remove TTY_DRIVER_MAGIC") [1], the magic field no longer exists in the tty_driver. Instead of printing the magic field, use the driver_name field.

[1] https://lore.kernel.org/linux-doc/20220907231109.htnfxzmr6wkuhn2v@tarta.nabijaczleweli.xyz/

Close #216
Close #217
commit b9b37fc3 (2023-08-10): kbleds: Replace magic with driver_name

Since the commit 5052df99d3bc ("tty: remove TTY_DRIVER_MAGIC") [1],
the magic field no longer exists in the tty_driver. Instead of
printing the magic field, use the driver_name field.

[1] https://lore.kernel.org/linux-doc/20220907231109.htnfxzmr6wkuhn2v@tarta.nabijaczleweli.xyz/

Close #216
Close #217