sysprog21/lkmpg #220: examples/sched.ko

tclee060219sat, 2023-08-24:
I am trying to figure out why the work_struct{} data values differ between INIT_WORK() and work_handler() executions, based on the same address of work_struct{}. Listed below are my runtime investigation and sched.c snippet.  

Your comments are appriciated.

My runtime investigation

Aug 24 10:44:04 raspberrypi kernel: sched_init()@28: queue == 0x0000000000000000
Aug 24 10:44:04 raspberrypi kernel: sched_init()@39: queue == 0x00000000723ff7a6
Aug 24 10:44:04 raspberrypi kernel: sched_init()@52: &work == 0x00000000e5cdb0ce
Aug 24 10:44:04 raspberrypi kernel: sched_init()@53: work.data == 0
Aug 24 10:44:04 raspberrypi kernel: sched_init()@55: work.data == 68719476704
Aug 24 10:44:04 raspberrypi kernel: work_handler()@22: data == 0x00000000e5cdb0ce
Aug 24 10:44:04 raspberrypi kernel: work_handler()@23: data->data == 64
Aug 24 10:44:09 raspberrypi kernel: sched_exit()@62

My examples/sched.c snippet
.....
#define fL __func__, __LINE__ // aug18fri23
.....
static void work_handler(struct work_struct *data)
{
    pr_info("%s()@%u: data == 0x%p\n", fL, data);
    pr_info("%s()@%u: data->data == %ld\n", fL, atomic_long_read(&data->data));
}

static int __init sched_init(void)
{
    pr_info("%s()@%u: queue == 0x%p\n", fL, queue);
    queue = alloc_workqueue("HELLOWORLD", WQ_UNBOUND, 1);
    pr_info("%s()@%u: queue == 0x%p\n", fL, queue);
    .....
    pr_info("%s()@%u: &work == 0x%p\n", fL, &work);
    pr_info("%s()@%u: work.data == %ld\n", fL, atomic_long_read(&work.data));
    INIT_WORK(&work, work_handler);
    pr_info("%s()@%u: work.data == %ld\n", fL, atomic_long_read(&work.data));
    .....
}
.....

jserv, 2023-08-24:
Thank you for bringing up the issue. To ensure effective communication and resolution, I recommend taking a moment to review the guidelines outlined in [Writing Good Bug Reports](https://github.com/rstudio/rstudio/wiki/Writing-Good-Bug-Reports). This will help us better understand and address the problem. Additionally, feel free to leverage ChatGPT for proofreading assistance, as it can be quite helpful in refining messages. Your efforts in improving the descriptions are greatly appreciated!

tclee060219sat, 2023-08-24:
Thank you for introducing ChatGPT and other web sites which will improve my communication skills to lkmpg. I will spend time on those guiddelines.

jserv, 2023-10-02:
Close due to being inactive.

closed 2023-10-02 by jserv

mechanicalamit, 2023-10-07:
data after INIT_WORK has a value from whatever it gets from memory.

Here is the call order on scheduling work: 
schedule_work()  -> queue_work() -> queue_work_on() -> __queue_work() -> insert_work() -> set_work_pwq() -> set_work_data()

[set_work_data()](https://elixir.bootlin.com/linux/v6.5.6/source/kernel/workqueue.c#L674) has the line of code below
`atomic_long_set(&work->data, data | flags | work_static(work));`

That's why data changes in between the INIT_WORK() and work_handler() within the same work structure. It is modified when the work is scheduled.