sysprog21/lkmpg #233: About /proc and try_module_get

m-sedl, 2023-09-23:
Hi! 
I read sections 7.0-7.1 about procfs and tried to run example procfs1.c. It's work, but I have some questions.

1. Why we don't use `try_module_get` and `module_put` in procfs module, but use in chardev module? There is this text, but I don't understand why we don’t get called when the file is opened or closed.
> Because we don’t get called when the file is opened or closed, there’s nowhere for us to put try_module_get and module_put in this module, and if the file is opened and then the module is removed, there’s no way to avoid the consequences.

2. I compiled procfs1.c and load module successfully. 
```shell
msedlyarskiy@my-dev ~/l/procfs1> sudo insmod procfs1.ko
msedlyarskiy@my-dev ~/l/procfs1> sudo cat /proc/helloworld
HelloWorld!
```

Then, I executed `cat` and got this dmesg output: 
```
[1326668.263173] procfs1: loading out-of-tree module taints kernel.
[1326668.263316] procfs1: module verification failed: signature and/or required key missing - tainting kernel
[1326668.263633] /proc/helloworld created
[1326686.364398] procfile read helloworld
[1326686.364412] copy_to_user failed
```

All right. But then I tried to open mounted proc file by python3.
```shell
msedlyarskiy@my-dev ~/l/procfs1> sudo python3
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> f = open('/proc/helloworld', "r", encoding="utf-8")
fish: Job 1, 'python3' terminated by signal SIGKILL (Forced quit)
```

Python3 failed and in dmesg I found some errors:
```
[1327018.190609] R13: 00005648566695a0 R14: 0000000000000001 R15: 0000000000000003
[1327018.191691]  </TASK>
[1327018.192044] Modules linked in: procfs1(OE) cpuid tls xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nf_tables nfnetlink br_netfilter bridge stp llc overlay intel_rapl_msr intel_rapl_common joydev nfit input_leds mac_hid serio_raw qemu_fw_cfg binfmt_misc sch_fq_codel dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua ipmi_devintf ipmi_msghandler msr pstore_blk ramoops pstore_zone reed_solomon efi_pstore ip_tables x_tables autofs4 btrfs blake2b_generic zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear hid_generic usbhid hid bochs drm_vram_helper drm_ttm_helper ttm drm_kms_helper crct10dif_pclmul syscopyarea sysfillrect sysimgblt crc32_pclmul fb_sys_fops ghash_clmulni_intel cec aesni_intel rc_core virtio_net net_failover crypto_simd i2c_i801 psmouse failover cryptd drm virtio_blk
[1327018.192107]  lpc_ich i2c_smbus
[1327018.204675] CR2: 0000000000000000
[1327018.205180] ---[ end trace 198b4be9390f4bd4 ]---
[1327018.205891] RIP: 0010:0x0
[1327018.206299] Code: Unable to access opcode bytes at RIP 0xffffffffffffffd6.
[1327018.212234] RSP: 0018:ff4bcacdc0e9be20 EFLAGS: 00010246
[1327018.212997] RAX: 0000000000000000 RBX: ff4148699643f780 RCX: 0000000000000001
[1327018.214029] RDX: 0000000000000001 RSI: 0000000000000000 RDI: ff4148699566d200
[1327018.215090] RBP: ff4bcacdc0e9be38 R08: 0000000000000003 R09: 0000000000000000
[1327018.216153] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000001
[1327018.217192] R13: 0000000000000000 R14: ffffffffffffffea R15: ff4148699566d200
[1327018.218200] FS:  00007f432a0531c0(0000) GS:ff41486bf3cc0000(0000) knlGS:0000000000000000
[1327018.219299] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[1327018.220111] CR2: ffffffffffffffd6 CR3: 000000010e5b0002 CR4: 0000000000771ee0
[1327018.221107] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[1327018.222086] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[1327018.223073] PKRU: 55555554
```

It's similar to trying executed nonexistent code, but I am not sure. I couldn't unload module because rmmod stucked.

Can you help me to understand  why can cat read proc file but python3 can't?

mechanicalamit, 2023-10-07:
Is the python3 crash reproducible 
`
Python 3.11.5 (main, Sep  2 2023, 14:16:33) [GCC 13.2.1 20230801] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> f = open('/proc/helloworld', "r", encoding="utf-8")
>>> f
<_io.TextIOWrapper name='/proc/helloworld' mode='r' encoding='utf-8'>
>>> f.read()
'HelloWorld!\n\x00'
>>> f.read()
''
>>> f.close()
>>> f = open('/proc/helloworld', "r", encoding="utf-8")
>>> f.write("Hello")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
io.UnsupportedOperation: not writable
`

jserv, 2024-01-30:
Close since the fix was landed.

closed 2024-01-30 by jserv