sysprog21/lkmpg #313: Improve readability of `devicemodel_remove()` version-dependent definition

EricccTaiwan, 2025-04-24:
The current implementation of `devicemodel_remove()` uses conditional compilation to switch the return type between `void` and `int`, as shown below:

```c
#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 11, 0) 
static void devicemodel_remove(struct platform_device *dev) 
#else 
static int devicemodel_remove(struct platform_device *dev) 
#endif 
{
    pr_info("devicemodel example removed\n");

    /* Your device removal code */

#if LINUX_VERSION_CODE < KERNEL_VERSION(6, 11, 0) 
    return 0; 
#endif
}
```

This approach works functionally, but it's not intuitive. The logic is split across multiple `#if`/`#endif` blocks, making it harder to read and maintain.
I propose replacing it with a clearer structure by splitting the entire function definition based on kernel version:

```c
#if LINUX_VERSION_CODE >= KERNEL_VERSION(6, 11, 0)
static void devicemodel_remove(struct platform_device *dev)
{
    pr_info("devicemodel example removed\n");
    /* Your device removal code */
}
#else
static int devicemodel_remove(struct platform_device *dev)
{
    pr_info("devicemodel example removed\n");
    /* Your device removal code */
    return 0;
}
#endif
```

This improves readability and maintainability by clearly separating concerns based on kernel version.

RafiulPaceProjects, 2025-05-18:
Your right it does look better!

jserv, 2025-05-18:
> This makes the logic easier to understand and reduces potential confusion for future readers. If this looks good, I can send a patch for it.

Please go ahead.

closed 2025-05-18 by jserv

EricccTaiwan, 2025-05-18:
> Your right it does look better!

Thanks.

PR #321 (2025-05-18): examples: Refactor device model with version-based split
The previous implementation of devicemodel_remove used conditional compilation to switch between void and int return types based on the kernel version. While functionally correct, the interleaved `#if/#endif` blocks made the logic harder to follow.

This refactor cleanly separates the two function definitions using a full `#if/#else` block around the entire function.

No functional behavior is changed.

Close: #313  
 <div id='description'>
<h3>Summary by Bito</h3>
This pull request refactors the device model implementation, enhancing th [... cut]
commit 38a991d3 (2025-05-18): examples: Refactor device model with version-based split

The previous implementation of devicemodel_remove used conditional
compilation to switch between void and int return types based on
the kernel version. While functionally correct, the interleaved
`#if/#endif` blocks made the logic harder to follow.

This refactor cleanly separates the two function definitions using
a full `#if/#else` block around the entire function.

No functional behavior is changed.

Close: #313